PicoPlus-devel/pico-infonesPlus #1: Try to implement Pico W support

fhoedemakers, 2023-01-04:


DoomHammer, 2023-02-18:
What is needed here? How much does Pico W differ from a standard Pico in terms of this project?

I assume we're not gonna use wireless features anyway, so it's just a matter of porting the existing features to a new board which should be fairly similar. What am I missing?

fhoedemakers, 2023-02-18:
I hoped it was that simple. 
First of all, the led cannot be used on the W. It blinks every 60 frames. For the pico W, in order to use the led, you have to  call cyw43_arch_init(). This raises an out of memory panic.  
I tried to bypass the led and not use it at all, but then it halts during initialisation of the DVI library.  Maybe also memory related, but this must be investigated further.

fhoedemakers, 2023-02-19:
Ok, i can confirm that the emulator also works on the Pico W. But without the blinking onboard led.
I don't know how anymore how i came to the conclusion that it would not work. I maybe had a faulty board.

closed 2023-02-19 by fhoedemakers

DoomHammer, 2023-02-19:
I think we can live without the blinking led :)