PicoPlus-devel/pico-infonesPlus #2: Support for the Hack Club Sprig

Vishram1123, 2023-01-22:
I am trying to get this program to work on the Hack Club Sprig [(schematic here)](https://github.com/hackclub/sprig/blob/main/docs/GROWING_A_SPRIG.md), and while I can flash the rom and load the menu (and it shows games), I cannot get the controls to work. This is expected, as you stated that this works only on the DualShock, DualSense, and the BUFFALO BGC-FC801. However, could you possibly use this schematic to add support for the Sprig?

fhoedemakers, 2023-01-22:
Unfortunately i don't think i can be of any help in this.

Since the emulator is build for HDMI output (including sound), the display and audio code should be rewritten for use with a TFT breakout board and a separate audio board. 
De Adafruit TFT ST7735 specified in your link has a resolution of  128x160  pixels. This is too low, since a resolution of 256x240 is needed,

Vishram1123, 2023-01-22:
Alright, thanks anyway

Vishram1123, 2023-04-24:
Do you think it would be possible to adapt your PicoSystem handheld version to work with this?

fhoedemakers, 2023-04-24:
Hi, but the resolution is still too low for that device. (128x160 pixels). The NES needs 256x240 pixels. SInce the PicoSystem also has a resolution of 240x240 pixels, i needed to cut-off 8 pixels to the left and 8 pixels to the right. This is somewhat acceptable. But for only 128 pixels screenwidth, this is way too short. Or do i miss something here?

Vishram1123, 2023-04-29:
You could downscale by getting the average of every other pixel in the x and y directions, so an image like 
[image]

fhoedemakers, 2023-05-03:
Sorry for the late response. Was quite busy. 
This could be an option with the PicoSystem handheld as a starting point, however i do not have the device at hand. Can you be of any help youtself in this?

Vishram1123, 2023-05-06:
Yes, I have the device and I can help (though I am also in the middle of final exams, so it might take a bit to respond); right now, the PicoSystem rom (expectedly) does nothing on my device, but it does show a blue light on the D1 led.