PicoPlus-devel/pico-infonesPlus #6: can't see memory card

Anarhist8BIT, 2023-02-04:
does not see a card with this board, is it possible to make it work with it?
[https://img.joomcdn.net/3a98c1d4841ae8d3a28e5b50fb603785311d8f76_original.jpeg](url)

fhoedemakers, 2023-02-04:
Can you specify the exact brand and product type of this SD breakout board? I cannot see this on the picture.

Anarhist8BIT, 2023-02-04:
Interfacing Catalex Micro SD Card Module with Arduino
Catalex Micro SD Card Adaptors

fhoedemakers, 2023-02-05:
Hi,
This specific breakout board needs 5V as input (VCC). Not sure if it works with 3,3V input. Please try to connect board's VCC to the Pico's VBUS pin (pin 40).  See breadboard image in README file where the HDMI board is connected to the same pin (Red wire going up).
Also use quality branded SD-cards, formatted as FAT32.
Hope this helps.

Anarhist8BIT, 2023-02-07:
> 
I connected directly 3.3 and 5v did not give any result, I tried different memory cards 128mb, 4gb.  16 GB.  wormat fat32.  did not help

fhoedemakers, 2023-02-07:
Did you also try another Pico? 

I ordered a Catalex SD card module myself and i will see if i can make it work.

Anarhist8BIT, 2023-02-07:
Tried 3 different ones, assembled from scratch, tried the same new slot.  didn't work out.

fhoedemakers, 2023-02-09:
Got it working. Please download [this file ](https://github.com/fhoedemakers/pico-infonesPlus/raw/main/assets/piconesPlusCatalex.uf2). Flash it to your Pico and test.
Make sure the SD card reader is connected to 5V. 3V does not work.
Appreciate your feedback.
Thanks!

[image]

Anarhist8BIT, 2023-02-09:
thanks a lot! you make me happier) everything worked

Anarhist8BIT, 2023-02-10:
sometimes an error about the map may still appear, sometimes the picture disappears during the game, and sometimes just a red screen.

fhoedemakers, 2023-02-12:
- From the readme:  Due to the Pico's memory limitations, not all games will work. Games not working will show a "Mapper n is unsupported." (n is a number). For example starting Castlevania III will show the "Mapper 5 is unsupported." message.

- I don't now why the picture sometimes dissapears/goes red.  Do all the games show this behaviour or certain ones? In the latter case, can you provide me some examples?

closed 2023-02-18 by fhoedemakers