PicoPlus-devel/pico-infonesPlus #7: out of memory panic when running emulator build with the latest pico sdk

fhoedemakers, 2023-02-19:
The emulator does not work with the latest version of the pico sdk. It raises an out of memory panic.

Workaround ifor now s using an older branch of the sdk. The last known working branch is pico_rand

You can clone this with

git clone https://github.com/raspberrypi/pico-sdk.git --branch pico_rand

fhoedemakers, 2023-06-21:
Thanks for the suggestion.
I removed the mapper and changed the build scripts. The program now runs fine built with the 1.5 SDK. 
As i can see there is no need for mapper 6 to be included.
Sound also still works.  (#21)
Also based on your comment, sound seemed to be working. Or was this using the 1.4 build?

fhoedemakers, 2023-06-21:
See attachment. Built with 1.5 for the PimoroniDV
[piconesPlus.zip](https://github.com/fhoedemakers/pico-infonesPlus/files/11819091/piconesPlus.zip)

kholia, 2023-06-21:
> Built with 1.5 for the PimoroniDV...

Yep - this build reproduces the problem (i.e. No HDMI sound) on the `Pimoroni Pico DV Demo Base` setup.

fhoedemakers, 2023-07-04:
Issue solved in 0.8 release