PicoPlus-devel/pico-infonesPlus #11: Adapt to VGA

tinspin, 2023-02-28:
Could this be adapted to the VGA board:

https://shop.pimoroni.com/products/pimoroni-pico-vga-demo-base

fhoedemakers, 2023-03-02:
I don't have the board myself. The graphics side  could be made working i guess.  Concerning audio  however,  this has to go through the DAC to the headphone jack and must be coded. The emulator takes up almost the whole memory and it will be a challenge to fit the additional audio processing into it.

tinspin, 2023-03-02:
Ah, true. I bought the HDMI board... will try to make it work with a USB - I2C NES controller.

Why can't you use the board USB power instead of the Pico USB?

fhoedemakers, 2023-03-02:
No the board usb power of the pimoroni dvi can be used without a problem.

fhoedemakers, 2023-03-04:
I was not aware this product existed. 

For a few weeks now, i'm experimenting to get it to work on a small LCD display ( Pimoroni Pico Display Pack 2). It  works and is playable, but the framerate is too slow.  The display cannot display the frames fast enough. I maybe have to use a smaller display  than this 320x240 one i have on hand.  
Also audio is still missing. I will try the Pico audio pack for this. Eventually, all this can be later adapted for use on the Pico System Handheld. But still a lot of hurdles to be taken.

[image]

tinspin, 2023-03-04:
Are you using both cores? If you move the screen framebuffer handling (and sound later) to the other core and overclock it?

fhoedemakers, 2023-03-04:
Yes the second core is used for screen updates. As far as i can tell, it is the display that just isn't fast enough and is the bottleneck here. So overclocking the cpu will not work. But i can be wrong, i'll still have to learn a lot.

fhoedemakers, 2023-03-05:
Closing this one, for Pico System see  [#12 ]

closed 2023-03-05 by fhoedemakers

dglaude, 2023-03-05:
About the frame rate, I believe the limitation might be the SPI connection that is a limiting factor for the FPS.
One way to increase the FPS is to do partial optimized screen update, but that require keeping the previous frame and compare, but it has a cost and not always possible.
Another way is to use screen with parallel connection, that require more pins to connect, you can see that on Adafruit PyPortal.

fhoedemakers, 2023-03-19:
@tinspin @dglaude Early release for the PicoSystem can be found here: https://github.com/fhoedemakers/PicoSystem_InfoNes
Unfortunately no sound in this release. Feedback appreaciated. :-)