PicoPlus-devel/pico-infonesPlus #12: Would it be possible to port to picosystem?

kefabean, 2023-03-05:
I would curious to understand whether it would be possible to port to pimoroni picosystem? would this be possible?

fhoedemakers, 2023-03-05:
I think it can be done. 

I'v already been working to get the emulator work on a similar display (Pico Display pack 2). Still work in progress, no sound, framerate too low.  Concerning the Pico System, it has no SD card, but 16MB flash. So there has to be a user friendly way to flash the system with roms. Maybe some kind of companion app for PC with a GUI. Also i'm not sure about the sound capabilities of the Pico System.

[image]

kefabean, 2023-03-18:
This looks amazing. Awesome work.I’m not sure I can be much use solving any thorny issues, but as my kids have a couple of picosystems, if there’s anything I can do to compile, test for picosystem let me know.

fhoedemakers, 2023-03-18:
Work in progress. See [here](https://twitter.com/frenskefrens/status/1637156593946046465?s=46&t=eco_RMZT-2pTUZjTmeRpNw)

fhoedemakers, 2023-03-19:
@kefabean Early release for the PicoSystem can be found here: https://github.com/fhoedemakers/PicoSystem_InfoNes
Unfortunately no sound in this release. Feedback appreaciated. :-)
[image]

kefabean, 2023-03-19:
So cool, thanks! will definitely give it a try and report back. Btw is it relatively straightforward to compile from source in case there’s a way I’m able to contribute?

fhoedemakers, 2023-03-19:
Yes should compile without any issues. Just let me know if you have any problems so I can help you out.

fhoedemakers, 2023-08-09:
Closed because project is ported to PicoSystem

closed 2023-08-09 by fhoedemakers