PicoPlus-devel/pico-infonesPlus #21: No HDMI sound when built with the latest Pico SDK (v1.5.0)

kholia, 2023-06-21:
After enabling the https://github.com/fhoedemakers/pico-infonesPlus/issues/7#issuecomment-1599126518 workaround, I can load the ROMs but there is no HDMI sound.

HDMI sound works fine when the emulator is built against Pico SDK v1.4.0 (an older version).

fhoedemakers, 2023-06-21:
Hi, 
As mentioned in #7 the sound still works for me on the 1.5 SDK.
Can you tell me what config you are using? Breadboard or Pimoroni DV? 
Also, If you soldered the headers yourself on the Pico, make sure you go over them again. I had a previous issue regarding no sound which was related to bad solder joints,

Please let me know.

Frank

kholia, 2023-06-21:
I am testing using the `Contra NES` ROM, in case it helps to debug the problem / user error.

fhoedemakers, 2023-06-21:
I see it now. 
Its related to the 1.5 usb driver. I tested using the breadboard config with a original NES controller. There are no issues with sound using this setup. 
However, when i connect a PS5 controller via USB, the sound issues obviously occurs. It continuously drops. 
Have to look into this further.

kholia, 2023-06-21:
https://github.com/phdussud/pico-dirtyJtag/commit/95df63df9ad1e27d02aaf02e73f8e94112fb508d#diff-7a03e16f217a545ebfe80ba35bf57431962395bef00a2252c12b5e67274decc6R205

```diff
  if (tx_buf != output_buffer)
  {
    tud_vendor_write(tx_buf, output_buffer - tx_buf);
+   tud_vendor_flush();
  }
```

This is not related but it seems that "flushing" of USB writes explicitly is now required in USB apps built with Pico SDK v1.5.0.

I couldn't locate such a code pattern in this repository but I just started looking.

fhoedemakers, 2023-06-22:
1. Strange: When waiting long enough in a game, the sound comes back and kind of works. 
2. Same problem in the original emulator port https://github.com/shuichitakano/pico-infones

fhoedemakers, 2023-06-26:
I asked the original developer of the port https://github.com/shuichitakano/pico-infones to look into this and he got sound working  again.  It is solved in the pico_lib submodule.
I will merge  the master branch asap when i'm done with other changes which i'll have to add too.
For now, you can make it work by updating the pico_lib submodule to the latest master branch and rebuild.

[image]

closed 2023-06-26 by fhoedemakers