PicoPlus-devel/pico-infonesPlus #26: Support another USB Gamepad

kpbicka, 2023-07-11:
Hi. How to add support USB gamepad with 
Product ID:	0x0011
 Vendor ID:	0x0079

Thanks

fhoedemakers, 2023-07-11:
What is the brand and type of this gamepad?

kpbicka, 2023-07-11:
> What is the brand and type of this gamepad?

with such identifiers, several names are produced, but specifically in my case it is called Defender[image]

kpbicka, 2023-07-16:
> What is the brand and type of this gamepad?

no any ideas?)

fhoedemakers, 2023-07-17:
Hi,
I don't have that specific gamepad, so i cannot test it.

Last week i tried adding different other USB gamepads, (8BitDO, XBox, Nintendo WII Switch, etc). Must of them were not detected at all by the TinyUsb driver, for those who were detected, the `tuh_hid_report_received_cb()` callback was not executed, so the data from the controller could not be read.
So even if i have that specific controller at hand, chances it will work are slim.

Maybe i'm doing something wrong here. The TinyUsb documentation is very limited.

I'll do some more testing when i have time for it.

Frank

hetii, 2023-08-08:
Hello dear.
I'm also in the middle of adding new pad for this project.
In my case it looks like below, and it's wireless.
[image]

This controller report itself under ubuntu like:
[image]

Then tinyusb driver also not enumerate it as a HID device. 
This is due the fact that in `hidh_open` of `src/class/hid/hid_host.c` tinnyusb library the bInterfaceClass is asserted and expected to have value 0x05. 
In my case for this controller it's 0xff Indicate that we have a deal with vendor specific controller.
Then discover this PR https://github.com/hathach/tinyusb/pull/1647 indicated that vendor specific driver is some sort in refactoring stage and not fully done yet in tinnyusb, but based on this PR I start creating own driver.

Meantime in this process I notice that pico sdk 1.4.0 use old tinnyusb library in version 0.12.0.
So now after a bit of work I successfully use tinnyusb in version 0.15.0 and code from above PR.
In this point my gamepad is just attached but additional work need to be done to handle endpoint and process data.

As I have blurry idea how this should be done finally I also discover example called [bare_api](https://github.com/hathach/tinyusb/blob/master/examples/host/bare_api/src/main.c)

After adding it to this project and ignoring checking of `if (desc_itf->bInterfaceClass == TUSB_CLASS_HID)` in `parse_config_descriptor` I was able to register endpoint of my controller and got events from it when pressing buttons ;)

Now about  `tusb_config.h`,  to use tinnyusb in version 0.15.0 you need to have `#define TUSB_OPT_MUTEX 1` otherwise pico hangs when new usb device is pluged.

Then `#define CFG_TUH_API_EDPT_XFER 1`  it required to use bare_api example.

Now sure why but when I plug my pad into pico, the idVendor and idProduct is has values 0x045e:0x028e instead of 054c:0cda, you can see that below
[image]

So my work is still in progress and I imagine that we can add support for regular keyboards for those who don't have pads and wanna to just test this project. 

I also wonder if all that usb enumeration parts can be handled by second core of pico as the screen blinks when new usb device is attached.

If you have any suggestion about my vendor specific pad or how you would see that implemented at the end (by bare api or vendor driver) please let me know and don't hesitate to contact me.

fhoedemakers, 2024-10-20:
Not sure if you are still working on this, but the 0x054c 0x0cda playstation classic controller (At least the original one) now works with the emulator

> Hello dear. I'm also in the middle of adding new pad for this project. In my case it looks like below, and it's wireless. ![obraz](https://private-user-images.githubusercontent.com/3405461/259080307-6755041c-651d-4825-b913-46d17d271570.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk0MTkwMTQsIm5iZiI6MTcyOTQxODcxNCwicGF0aCI6Ii8zNDA1NDYxLzI1OTA4MDMwNy02NzU1MDQxYy02NTFkLTQ4MjUtYjkxMy00NmQxN2QyNzE1NzAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjBUMTAwNTE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODVkYjM2NGZhYmIzZmNiZjNkYWFmMGFkZTcxMDA5Mzc5MGNiMGYxMzNhNTJkMzg1ZjcwY2Q5M2I4N2ZlMDdhNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.bX1HrnnlDw--nqgohPoIHkBWmsk6IufnUE8-raf9AgU)
> 
> This controller report itself under ubuntu like: ![obraz](https://private-user-images.githubusercontent.com/3405461/259080905-06e10a01-f22f-4b3a-8f8a-c898cabbeec2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk0MTkwMTQsIm5iZiI6MTcyOTQxODcxNCwicGF0aCI6Ii8zNDA1NDYxLzI1OTA4MDkwNS0wNmUxMGEwMS1mMjJmLTRiM2EtOGY4YS1jODk4Y2FiYmVlYzIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjBUMTAwNTE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2JiODQ1YThjNDQ3MWRkZDNjZDA5NDY3MTk4ZDZlNmRhOGM4ZDM0NjUzMTk0ZjQ2ZWMzOWIxMWQ1ZDU0OTg3ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.cpDO6j3zOFycqHEYlEpTOXX8NaTAWr42-kb5cYNPi7Q)
> 
> Then tinyusb driver also not enumerate it as a HID device. This is due the fact that in `hidh_open` of `src/class/hid/hid_host.c` tinnyusb library the bInterfaceClass is asserted and expected to have value 0x05. In my case for this controller it's 0xff Indicate that we have a deal with vendor specific controller. Then discover this PR [hathach/tinyusb#1647](https://github.com/hathach/tinyusb/pull/1647) indicated that vendor specific driver is some sort in refactoring stage and not fully done yet in tinnyusb, but based on this PR I start creating own driver.
> 
> Meantime in this process I notice that pico sdk 1.4.0 use old tinnyusb library in version 0.12.0. So now after a bit of work I successfully use tinnyusb in version 0.15.0 and code from above PR. In this point my gamepad is just attached but additional work need to be done to handle endpoint and process data.
> 
> As I have blurry idea how this should be done finally I also discover example called [bare_api](https://github.com/hathach/tinyusb/blob/master/examples/host/bare_api/src/main.c)
> 
> After adding it to this project and ignoring checking of `if (desc_itf->bInterfaceClass == TUSB_CLASS_HID)` in `parse_config_descriptor` I was able to register endpoint of my controller and got events from it when pressing buttons ;)
> 
> Now about `tusb_config.h`. To use tinnyusb in version 0.15.0 you need to have `#define TUSB_OPT_MUTEX 1` otherwise pico hangs when new usb device is plugged.
> 
> Then `#define CFG_TUH_API_EDPT_XFER 1` is required to use bare_api example.
> 
> Now, i'm not sure why, but when I plug my pad into pico, the idVendor and idProduct is have values 0x045e:0x028e instead of 054c:0cda, you can see that below: ![obraz](https://private-user-images.githubusercontent.com/3405461/259086966-74feac42-8146-4b2b-9598-abeba591de37.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk0MTkwMTQsIm5iZiI6MTcyOTQxODcxNCwicGF0aCI6Ii8zNDA1NDYxLzI1OTA4Njk2Ni03NGZlYWM0Mi04MTQ2LTRiMmItOTU5OC1hYmViYTU5MWRlMzcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjBUMTAwNTE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NGMwYjAxNjNlY2ZmMzg0MGFjYjc1ZThlZDE1NTU2NmQxOWI5Yjc0MzViN2NjZDA2ZjcyMTcyODE4MzcwOWI0YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.5pWVihDDobJ3IHUx-fW7cF2xtkBnilYxW_j9ht2yOLE)
> 
> So my work is still in progress and I imagine that we can add support for regular keyboards for those who don't have pads and wanna to just test this project.
> 
> I also wonder if all that usb enumeration parts can be handled by second core of pico as the screen blinks when new usb device is attached.
> 
> If you have any suggestion about my vendor specific pad or how would you see that implemented at the end (by bare api or vendor driver) please let me know and don't hesitate to contact me.
> 
> Best Regards.

fhoedemakers, 2024-10-20:
> Hi. How to add support USB gamepad with Product ID: 0x0011 Vendor ID: 0x0079
> 
> Thanks

Not sure if you are still interested, but i can send you a test program to see wheter this controller works

closed 2024-10-24 by fhoedemakers