PicoPlus-devel/pico-infonesPlus #29: Add analog audio output

Sky-Jumper, 2023-09-10:
Is it possible to add analog audio output to the free rp2040 pins, for connection to the speaker system, not via hdmi?

fhoedemakers, 2023-09-14:
Thanks for your comment. I'll look into this, cannot guarantee anything however.

Frank

fhoedemakers, 2025-05-31:
@Layer812
I added SPI audio for use with the DAC on the Pimoroni Pico DV Demo base.  Below a zip file with .uf2 binaries for Pico and Pico 2

[piconesPlusPimoroniDV.zip](https://github.com/user-attachments/files/20533279/piconesPlusPimoroniDV.zip)

The changes are done in these two branches:

https://github.com/fhoedemakers/pico-infonesPlus/tree/i2s_audio
https://github.com/fhoedemakers/pico_shared/tree/i2s_audio

Let me know how this works.

fhoedemakers, 2025-06-01:
Thanks for testing! There is some crackling in the noise sometimes due to dropped samples i think.

closed 2026-02-14 by fhoedemakers