PicoPlus-devel/pico-infonesPlus #31: SD card mount error: 3

OldMate6288, 2023-11-23:
Hi! I got a few of your PCBs made and put it together with the recommended components. Everything seems to be working but I keep getting mount error: 3 on boot. I am using a 16gb SDHC card (FAT32), also tried using a 64gb SDXC card (formatted FAT32 with GUIformat and tried making a 1gb partition but that threw mount error: 1). Looked through the main code too but I couldn't find what the error codes were. 

I am stuck as to what the issue is, my soldering seems to be fine but I will attach pics of everything if you want to double check.

[image]
[image]
[image]
[image]
[image]

fhoedemakers, 2023-11-23:
Hi, 
Sorry to hear it is not working for you. 
Make sure you installed the latest release for this configuration: https://github.com/fhoedemakers/pico-infonesPlus/releases/download/v0.8/piconesPlusAdaFruitDVISD.uf2 
Also i had issues using cheap SD cards. Make sure you use a card from a well known brand.
Can you also redo the soldering of the SD breakout at the back of the board? Looks like the D1 pin is not fully soldered.

Hope this helps.

Frank

OldMate6288, 2023-11-23:
Thank you! Turns out it was my soldering, must've missed that pin the first time 🤦

Sorry to open an issue just for this, will be more careful in future.

closed 2023-11-23 by OldMate6288

fhoedemakers, 2023-11-23:
No problem, glad it works! Enjoy!