PicoPlus-devel/pico-infonesPlus #32: Boots to red screen

abeisgoat, 2023-12-19:
Hello @fhoedemakers thanks for making this sick project.

I just got my Pico DV in and tried to boot this up but I just get a solid red screen out the HDMI.

I believe I've set up my SD correct and whatnot.

Any idea what may be the issue? I've loaded up the PicoDVI examples without issue so I believe at least the video part of the wiring is correct!

Thanks in advance,
Abe

fhoedemakers, 2023-12-19:
Hi,
Sorry to hear it is not working for you. The red screen indicates a program crash. On the original Pico (not the W). There is a blinking led indicating normal operation. 

As i read correctly, you are using the Pimoroni DV board. ( https://shop.pimoroni.com/products/pimoroni-pico-dv-demo-base?variant=39494203998291 )

- Make sure you use the correct .uf2 file to flash onto the Pico. For the Pimoroni DV, use the latest one from the releases page https://github.com/fhoedemakers/pico-infonesPlus/releases/download/v0.8/piconesPlusPimoroniDV.uf2  If you followed the [Adafruit Guide](https://learn.adafruit.com/nes-emulator-for-rp2040-dvi-boards/using-pico-dv-demo-base), the .uf2 they offered for download does not work. I brought this to their attention and they said they fixed it, but i'm not sure they really did. So, use the release from the link above.
- Check the solder joints. When you have soldered the headers yourself onto the Pico, please recheck and reflow them.
- Try another Pico. Also use the original hardware. Not sure it works at all with clone or third party boards.

The software writes al lot of debug info to the serial terminal over GPIO0 and GPIO1. See  section 4.5 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf how to wire this up.
With that output, we probably can see where it crashes.

Hope this helps to get it solved.

Frank

abeisgoat, 2023-12-19:
Thank you for your help - it may have been an issue with the image I flashed, but upon flashing the one you linked it worked perfectly. I have featured this project in a [short on my Youtube channel](https://youtube.com/shorts/-m4QfUTgC7w?si=l9CxBSgD6cCe0vMY). Thanks for the help!

closed 2023-12-19 by abeisgoat