PicoPlus-devel/pico-infonesPlus #34: Add support for Waveshare RP2040-PiZero

lewilou22, 2024-03-06:
Hi, 

I have the Waveshare RP2040-PiZero. It has 2 USBC ports, hdmi, SD slot & battery input.

Is there a way to add support for this board ?

TIA e

dglaude, 2024-03-06:
That board is also supported for https://github.com/jfedor2/hid-remapper because of the two USB-C  (one for power and one for USB Host with an adapter)... so I plan to get one.

The mini(?) HDMI is a nice addition that fit emulator, retro-computer, ...

fhoedemakers, 2024-03-07:
Hi,
Will look into this. and see if it can work.
At the moment i don't have the board at hand.  So it may take a while. 
Frank

lewilou22, 2024-03-09:
If you need me to test anything with my board just let me know

fhoedemakers, 2024-03-21:
Hi,

Received the board. 

First try is the vanilla emulator from https://github.com/shuichitakano/pico-infones. This in order to get my troubleshooting right.
The emulator works, with sound and image displayed, but as soon as i plug in a usb controller, the sound stops and you only hear crackling noises.  I attached the .uf2 file as a zip so you can try it yourself. Maybe i have a faulty board.

You need to use picotool to flash a rom to the board:

> picotool load foo.nes -t bin -o 0x10080000

[picones.zip](https://github.com/fhoedemakers/pico-infonesPlus/files/14696875/picones.zip)

Let me know what your results are.

lewilou22, 2024-03-21:
I will take a look at this when i get chance !

fhoedemakers, 2024-03-22:
Ok, my bad. I used an outdated DVI driver. Must have known this, beacuse i stumbled into exact the same issue last year. 
Included a zipped uf2 file with working sound for WaveShare RP2040.
[picones.zip](https://github.com/fhoedemakers/pico-infonesPlus/files/14720153/picones.zip)
Up next: Get the SD card working. I'll keep you posted.

Frank

fhoedemakers, 2024-03-22:
Below a complete working version with menu and sound for  WaveShare RP2040-PiZero
[piconesPlus.zip](https://github.com/fhoedemakers/pico-infonesPlus/files/14722614/piconesPlus.zip)

Only works with usb controller now. Will look into it to get it to work also with the original NES gamepad.

Looking forward to your feedback.

Frank

lewilou22, 2024-03-22:
Awesome! I am at work right now, but I got a chance to load it before I left, it worked from what I could see! The menu came on and the Rom loading screen appreared

dglaude, 2024-03-22:
With this version:

> Below a complete working version with menu and sound for WaveShare RP2040-PiZero [piconesPlus.zip](https://github.com/fhoedemakers/pico-infonesPlus/files/14722614/piconesPlus.zip)
(I had nothing with previous version)

I have a video signal out, and it complain about the SD card missing.
(same behavior on my two RP2040-PiZero)
Is this a version were we just put the ROM on an SD card?

Not sure I have a compatible USB joystick around.

fhoedemakers, 2024-03-22:
> With this version:
>
>
> I have a video signal out, and it complain about the SD card missing. (same behavior on my two RP2040-PiZero) Is this a version were we just put the ROM on an SD card?
>
> Not sure [...]

Yes. Make sure SD card is FAT32 formatted. Rom files must have the .nes extension. I use a 16GB SD card for testing.

lewilou22, 2024-03-22:
[image]

Working with sound and PS4 controller on HDMI

fhoedemakers, 2024-03-22:
> [image]
> 
> 
> 
> 
> 
> Working with sound and PS4 controller on HDMI 

Great! Glad it works.

lewilou22, 2024-03-22:
Thank you!! :) 

I plan on making a micro NES with 3D NES shell files (https://www.printables.com/model/132150-nes-frontloader-replica-shell) & scaling down to fit the WaveShare RP2040-PiZero :)

closed 2024-03-24 by fhoedemakers

dglaude, 2024-03-24:
> I plan on making a micro NES with 3D NES shell files (https://www.printables.com/model/132150-nes-frontloader-replica-shell) & scaling down to fit the WaveShare RP2040-PiZero :)

Maybe a NES controller shaped best fit the PiZero size (and you find 3D files for that).

PS: I was able to test the last release in this issue, reading the SD card and displaying on HDMI was OK, but I could not make my joystick (PS4) to work. I need a kind of USB OTG cable, with USB-C on one side and USB-A host on the other side.

fhoedemakers, 2024-03-24:
When using the PS4 or PS5 controller, you need an USB-C Y cable like this [one](https://a.co/d/dVY8X1Z).  With this, you can connect an USB-C power adapter and the controller. I tested this and this worked. (both PS4 and PS5 controllers)

I also do not recommend to power the board from the PIO-USB-C connector.  Instead connect your power supply to the port labelled USB. I had some weird issues when power supply was connected to PIO-USB.

fhoedemakers, 2024-03-31:
If you are interested: I just released a port for Sega Master System emulator. Still work in process, but it runs decently well.

[https://github.com/fhoedemakers/pico-smsplus](https://github.com/fhoedemakers/pico-smsplus)

PR #35 (2024-03-24): Support for Waveshare RP2040-PiZero added
[Added support for Waveshare RP2040-PiZero development board ](https://www.waveshare.com/rp2040-pizero.htm)( closes #34 )