PicoPlus-devel/pico-infonesPlus #39: waveshare rp2040 pi zero sdcard error

matteius, 2024-04-26:
Things I have tried:
* a 128 MB and a 128 GB sd card
* Tried formatting to fat32, then fat16, then also tried a 16GB partition on the 128 as fat 32 -- all the same result.
* I tried latest distribution for this device; I tried two versions back when it was first introduced, same result.
* I do have a couple roms on the sd card when I try this, I've also tried it without.   They have the .nes extension.

[image]

fhoedemakers, 2024-04-26:
Hi,

I'm sorry it does not work for you. Not sure what the problem can be.

The latest release for the RP2040-PIZero is verified to work (https://github.com/fhoedemakers/pico-infonesPlus/issues/34).  
I again downloaded piconesPlusWsRP2040PiZero.uf2 from the [release page](https://github.com/fhoedemakers/pico-infonesPlus/releases/latest) and flashed onto my Waveshare device.  Works fine.

SD card has to be formatted with FAT32. But you did this.

Error 13 means  (13) There is no valid FAT volume

Can you try the Waveshare sample code for SD card access  https://www.waveshare.com/wiki/RP2040-PiZero#Demo_3 to see if you can access the sd-card with the sample code? 

Hope it works out for you. Let me know.

Frank

fhoedemakers, 2024-05-09:
Were you able to get it working?

matteius, 2024-05-09:
@fhoedemakers I have not been able to -- is there a specific sd card you would recommend getting to try against?   I spent a bit of time diffing the sdcard drivers from the demo code and there were some differences-- I was able to compile the demo code but not really sure how to test it as the sdcard test doesn't do anything with the display.  Ran out of time after that ...

fhoedemakers, 2024-05-09:
I use some generic brand SD cards. Maybe the size of the card matters. I used 4, 16 or 32GB cards without any problem. Larger size cards maybe a problem for the fatfs driver i use in the project.

matteius, 2024-05-13:
I bought a couple of these and they came today, first one worked:  https://www.amazon.com/dp/B09X7C7LL1?psc=1&ref=ppx_yo2ov_dt_b_product_details

closed 2024-05-13 by matteius

fhoedemakers, 2024-05-13:
Great to read it works now. Have fun!

Frank

fhoedemakers, 2025-01-05:
Can be the usb-c power supply you are using has a too low amperage. I think 2 amps will do fine. This is a wild guess however and is your power supply ok.

lewilou22, 2025-01-06:
I will do a few more tests, I used USB C block with at least 2.1 amps.

The first 32 GIG card had "sd card mounting error 3"

The second 32 Gig SD card ( brand name card)  Worked with out issues