PicoPlus-devel/pico-infonesPlus #40: Rp2040-zero waveshare board psu

actraiser30, 2024-06-19:
Hello 

Probably a dump question but could not find the answer elsewhere 

How do i power the board ??

What usb c port I use to give power??

And how much power I give??

Is 5v 1amp ok??

Thanks

fhoedemakers, 2024-06-19:
Hi,

See image below. An USB hub will not work. If you want to connect a controller, you need the splitter cable [like this one](https://a.co/d/gKXWxHM) so you can connect power and the controller (Note: only Ps4 or Ps5 controllers will work). A normal USB-c phone charger will work, so i think 1amp will work.

[image]

actraiser30, 2024-06-19:
Thanks a lot

I cannot find this splitter cable locally

And to order it online will take months to arrive 
Can I use the second pio-usb port for the joypad or the keyboard??

Thanks

fhoedemakers, 2024-06-19:
Keyboard will not work. I tried to implement different types of usb controllers, but only Ps4 and Ps5 seem to work.
I think the USB-PIO can be used for powering the board, but i'm not sure.  You cannot use usb-pio for the controller, controller must be connected to the other usb-c port.

Sorry for that.

actraiser30, 2024-06-19:
No problem I'm sure you doing the best you can,so I cannot really use it coz I dont have the needed ps4,5 controller,the keyboard will not work,a hub will not work, and the splitter cable is not available. Haha 😆 nevermind

actraiser30, 2024-06-19:
I have the N8 Everdrive if you know it,it has an FPGA inside and your emulator works exactly like it, it reads the game rom from the card and write it to flash rom.

fhoedemakers, 2024-06-19:
I'll close the issue now. Let me know if you need more help.

closed 2024-06-19 by fhoedemakers