PicoPlus-devel/pico-infonesPlus #41: Dual player support request

airviolence, 2024-07-04:
Hi, I don't seem to find a way to support player2. Can you add support for player2 for two player games?

fhoedemakers, 2024-07-05:
Hi,

I have to look into this. If I remember correctly, I never succeeded connecting an usb hub, but I could be wrong. 

Please allow me some time to look into this. 

Frank

GizmoTheGreen, 2024-07-22:
is there enough free gpio for two NES controllers? I was thinking I could used an esp32 with BlueRetro to interface bluetooth capable controllers, and the emulator will just see it as normal NES controllers.

the blueretro pinout for the NES is a little confusing imho cause it has a levelshifter for NES 5v while the esp32 is 3.3v, and some pullups for PAL NES. so I think hookup to the pi pico will be easier since it's 3.3v?

fhoedemakers, 2024-07-26:
> is there enough free gpio for two NES controllers? I was thinking I could used an esp32 with BlueRetro to interface bluetooth capable controllers, and the emulator will just see it as normal NES controllers.
> 
> the blueretro pinout for the NES is a little confusing imho cause it has a levelshifter for NES 5v while the esp32 is 3.3v, and some pullups for PAL NES. so I think hookup to the pi pico will be easier since it's 3.3v?
> 
> (there is also a fork of blueretro that added I think i2c to communicate the button states, this was to interface a at32u4 to give it USB interface to a OG xbox. this could be another way to hookup a esp32 running blueretro with fewer wires, but would probably need more work on your end.)

Sorry for the late reply. Still recovering from surgery.
The NES controller support is written bij AdaFruit using pio. Have to look into this wheter it is possible to add a second controller, but don't hold your breath.

updawg, 2024-08-09:
Any plans on looking into the recently annouce pico 2 - it adds more gpio I believe. Regardless, thanks for your awesome work on this project!

fhoedemakers, 2024-08-09:
> Any plans on looking into the recently annouce pico 2 - it adds more gpio I believe. Regardless, thanks for your awesome work on this project!

Thanks!

Yes, will certainly do as soon i can purchase a Pico 2 board.

Frank

fhoedemakers, 2024-08-11:
@GizmoTheGreen  @airviolence 
Added second NES controller support. Did not test it with blueretro, but works with two traditional NES controllers. 
Don't know what config you are using, I included the .uf2 file for the BreadBoard/PCB version.
Second controller must be hooked up as follows with this build:

- CLK      GPIO9
- DATA   GPIO10
- LATCH GPIO11
- 3V
- GND

Note: For the PCB variant, you need to solder to the correct GPIO pins.

You can view the code in the development branch

Let me know if this is the configuration you can use or you need a .uf2 file  for the Adafruit Feather DVI, WaveShare RP2040 or Pimoroni Pico DV config.
Nevertheless, i will add second NES controller for these configurations soon. (Not sure about the Pimoroni Pico DV however)

Concerning second USB controller. I can connect a second controller using an USB hub and a OTG-Y cable. However the USB driver only detects one controller. It never detects the second one. Not sure how to proceed here.

Please let me know.

[piconesPlusAdaFruitDVISD.zip](https://github.com/user-attachments/files/16572973/piconesPlusAdaFruitDVISD.zip)

GizmoTheGreen, 2024-08-11:
I got the waveshare board for simplicity, it appears to work with blueretro but the key mapping is wrong. 
not sure if the error lies with blueretro or waveshare build (for example pressing up on my synced wii u pro controller gives me Dpad down in the emulator, and right is A, minus(select) is Dpad Up. 
they're mapped 1:1 in the blueretro config as in wii u dpad right should make NES dpad right.
I was going to note the errors and make a custom map to compensate before posting screenshot of the "corrected" mapping here later but you beat me to it with second player support so now I want to try it too. I'll make a full report within the coming week.

fhoedemakers, 2024-08-11:
Thanks for coming back.

Did not build for waveshare yet. I'll look into this tomorrow and send and updated .uf2

Frank

fhoedemakers, 2024-08-12:
Hi,

Below the .uf2  for WaveShare RP2040-PiZero with support for second  NES controller  port .

This is the GPIO Pin Layout:

- CLK GPIO10
- LATCH GPIO11
- DATA GPIO12
- 3V
- GND

Tested with original NES controllers.

[piconesPlusWsRP2040PiZero.zip](https://github.com/user-attachments/files/16587940/piconesPlusWsRP2040PiZero.zip)

fhoedemakers, 2024-08-18:
I will close this issue as the two player feature will be merged to the main branch. 

Feel free however to post any comments regarding blueretro.

Two player using two USB controllers is still not possible however. Must use two NES controller, or an USB and a NES controller.

PR #49 (2024-08-18): Added support for two player games by adding a second NES controller port
closes #41
closed 2024-08-18 by fhoedemakers