PicoPlus-devel/pico-infonesPlus #42: Save screen modes between games?

DynaMight1124, 2024-07-05:
Hi

Firstly thanks for the amazing project. I'd imagine 10 year old me would never believe you'd be able to run NES games on device thats half the price of a cup of coffee :)

One question I did have was about the screen modes, it defaults to stretched with scanlines, this isnt my preferred mode. I know you can change it quite easily in game, however it reverts back whenever a new game is started.

Would it be possible to remember the last selected screen mode either stored in flash or via a config file on the SD?

Thank you.

fhoedemakers, 2024-07-05:
Hi,

Sure. I can look into this. Please give me some time to implement this. 
Since I have some medical issues right now, it can take a few weeks.

Frank

DynaMight1124, 2024-07-05:
Thank you for getting back to me, of course take your time, health always comes first. Hope everything is ok.

Thanks

fhoedemakers, 2024-08-03:
Hi,

I finally got some time to work on this request. The last chosen screen mode is saved to /settings.dat on the SD card. I did not choose to write to flash since that causes too many stability issues.

When changing the screen mode (Select+Up or Select+Down), the changes are immediately saved. Therefore, you may see some red screen flicker due to the delay of writing to the SD card.

As a bonus, the last used position in the ROM selection menu is also saved to /settings.dat. 

Below  a zip file containing builds for all the supported configurations. You can also build from the development branch if you prefer.

Please let me know if this suits your needs so that I can publish it as a new release.

[piconesPlus.zip](https://github.com/user-attachments/files/16480787/piconesPlus.zip)

Frank

DynaMight1124, 2024-08-03:
Hi Frank

Thank you, I've tested the new release and works perfectly. Remembers my previously selected screen mode and like you've also added, the menu selection too.

I've rebooted, chosen different games, reset and so far its worked perfectly every time!

Thanks
Gavin

fhoedemakers, 2024-08-03:
Hi Gavin,

Great to see it works for you. I'll do some more testing myself. (I need to test every possible configuration.)

When everything works fine, I'll merge the changes into the new branch and create a new release.

Please allow me some time to work through this.

Best regards,

Frank

PR #47 (2024-08-04): Development
- Screenmode is retained in /settings.dat Closes #42 
- Last chosen menu item is remembererd. Also saved to /settings.dat Closes #46 
- Github Action is now self-hosted
- Latest release of https://github.com/shuichitakano/pico_lib/tree/master is used to prevent compiler errors in recent versions of gcc.
closed 2024-08-04 by fhoedemakers