PicoPlus-devel/pico-infonesPlus #43: Platted or non-platted holes on PCB

albank1, 2024-07-05:
Hi

Love this project. I have had an issue with uploading the zip gerber files  on https://learn.adafruit.com/nes-emulator-for-rp2040-dvi-boards/using-pico-breakouts to PCBWay. 

Their engineers came back with the following:
As shown in attached pics, pls confirm the specified four holes should be produced as plated or non-plated?
If they are plated, do you need us add openings on both sides to expose them?
If they are non-plated, we need cut 0.2mm copper around the holes. Is that acceptable for you?

I have gone with non-plated but not sure if this is right
[image]
[image]

Thanks

Alban

fhoedemakers, 2024-07-05:
Hi Alban,

The 4 points indicate the position of the Pico USB connector. No holes are needed, so non-plated is ok.

Let me know if this answer is sufficient for you, so I or you can close the issue.

Best regards,

Frank

albank1, 2024-07-05:
Many thanks for the clarification

closed 2024-07-05 by albank1