PicoPlus-devel/pico-infonesPlus #48: Add second NES controller port to PCB design

fhoedemakers, 2024-08-18:
As of v0.13, two player games can now be played. However the PCB is not designed for an extra NES controller port. (Although it is easy to add a second port anyway).

It would be nice to add a second controller port to the design, so no extra wires need to be soldered onto the Pico.

[image]

DynaMight1124, 2024-08-20:
It would be great if the PCB could be updated to remove the SNES port and replace with a second NES port for 2 player. I'd deffo build one and update my 3D Printed case. If someone does do that, I would also suggest that NES controller ports are moved so they are flush with the edge of the PCB as will make a case cleaner, also same with the Pico and SD Card board.

P.S. Not sure if you've seen, but looks like Aliexpress have cloned this and most likely using your software: https://www.aliexpress.com/item/1005007466998896.html

fhoedemakers, 2024-08-21:
Hi,

I reached out to the designer of the PCB. I hope he has time for it.

Concerning the AlieExpress clones. I see a lot of Chinese vendors selling those clones,   They do not reference the GitHub repo, so i guess the emulator is pre-flashed. Sigh...

Frank

fhoedemakers, 2024-08-28:
@DynaMight1124 just a small update. I got a response from the designer of the PCB. He will change the design to add the second controller.

DynaMight1124, 2024-08-28:
Ah nice one, I'll order some new boards when they are released

fhoedemakers, 2024-08-31:
This is the new design. Will now be tested. If all works fine, I'll send you/upload the gerber files. 

[image]

DynaMight1124, 2024-09-03:
Looks good to me, happy to see the controller ports closer to the edge of the PCB. Hopefully the design works as intended, I'll orders some boards as soon as released.

fhoedemakers, 2024-09-06:
@DynaMight1124 Got the boards today. I assembled one and it worked just fine. Below the gerber files.
[pico_nes_v2.0.zip](https://github.com/user-attachments/files/16910623/pico_nes_v2.0.zip)

I will publish the design to my repository tomorrow. (PCB branch) Still need to make changes to the README though. 

Good luck with your case design.

[image]

jedgarpark, 2024-09-06:
@DynaMight1124 your v1 case looks great, are you thinking of updating it for two player board I did?

jedgarpark, 2024-09-06:
@DynaMight1124 I moved the ports to the edge as you noted, LMK if there are any other changes you'd suggest to better support the case design.

DynaMight1124, 2024-09-06:
@fhoedemakers I was snooping a little earlier and saw the PCB branch plus new gerbers so already ordered the new boards a few hours ago. Cheapest method from JLCPCB so probably be 10-14 days before I get them, now wondering whether to use a Pico2.

@jedgarpark Yeah I fully intend to update the case for the new PCB with 2 ports. I'll possibly need to remove the fake cartridge flap so may lose a little of the aesthetics but I'll probably have a play over the next week or so, see what works visually. It wasnt originally intending to make the design so NES like but the controller port like lined up so perfectly, probably spent twice as long adding the NES style details than the initial case :)  Thank you for moving the ports and the overall PCB design.

fhoedemakers, 2024-09-07:
I merged the PCB branch with the main branch. The new PCB design can be downloaded from the [latest release](https://github.com/fhoedemakers/pico-infonesPlus/releases/latest) (v0.17)

I modified the [README](https://github.com/fhoedemakers/pico-infonesPlus#pcb-with-raspberry-pi-pico-or-pico-2) with the latest info and mentioned that @DynaMight1124  is working on a new to be released design.

I keep this issue open for further discussion on the case and pcb design.

Thanks @jedgarpark and @DynaMight1124 for the awesome work done!

DynaMight1124, 2024-09-07:
Been working on something new this afternoon. Mostly there! Uses the Waveshare RP2040-PiZero

[image]

fhoedemakers, 2024-09-07:
> Been working on something new this afternoon. Mostly there! Uses the Waveshare RP2040-PiZero
> 
> 
> 
> [image]
> 
> 

Cool! Looks awesome.

fhoedemakers, 2024-09-13:
John has made a [video](https://www.youtube.com/live/2uxM8TRU9Mk?si=OqpuNxYjpiFh_aHU&t=1005)  on the Adafruit YouTube channel. He also mentioned your 3D-design @DynaMight1124

DynaMight1124, 2024-09-13:
Thats awesome. Its a great project, happy to add a little bit. 

PCB's should be here end of next week. I'm aiming to have a 1 player and 2 player version, one similar to the current design (maybe identical if it fits!) with more NES like appearance and then the 2 player version with as much as I can get but I think the cartridge port and probably power buttons may need to go. See how it pans out.

jedgarpark, 2024-09-13:
Looking forward to your models @DynaMight1124 ! sorry, I put the P2 controller port on the right, so I don't think that'll work as the sole controller for a single player case -- it seems a controller plugged into P2 can pick and launch a game, but not actually control it. I tested on Super Mario Bros. so maybe this isn't always true.

fhoedemakers, 2024-09-13:
I would stick to one design based on pcb V2.0 with nes P1 on the left and P2 to the right. Maybe with a removable cover on an empty P2 port for those who only want to play 1-player games.
For owners of the original PCB I can always refer to the original case design.

Concerning the controllers: the menu listens to any controller, whether it is USB, nes or Wii. In emulator, I cannot determine which nes controller is connected and use that for P1. I can only detect whether an USB controller is connected. Therefore, when an usb controller is connected, this will always be P1, any connected nes controller (port 1 or 2), or Wii will be P2.

DynaMight1124, 2024-09-13:
Ah yes, forgot about that. It is normal for P1 to be on the left so it is the correct setup. It just slipped my mind.

I'll design v2.0 PCB for both ports, but as 0.2 PCB is still on the repo, I can always leave the original design to cover that incase anyone really only wants 1 player. 

Personally I'll probably never use 2 player but its nice to have. I was surprised how good the AliExpress aftermarket NES controllers felt, felt very responsive and really well priced. Its hard to compare to original controllers as they're all like 35-40 years old at this point so probably would feel a little softer anyway.

DynaMight1124, 2024-09-21:
I've updated the design around v2.0 PCB, only needed slight adjustments so didnt take too long. I tried some different filament to try to mimic old yellowed plastic, looks a little better in person than the pictures :) 

Updated the files on Thingiverse: https://www.thingiverse.com/thing:6689537
[image]

fhoedemakers, 2024-09-21:
Great work! Will update the readme tomorrow!
Thanks!

fhoedemakers, 2024-09-21:
Just a side note. I'm not sure you soldered the ground on the pico's debug side. The traces of the sd and hdmi are connected to this. 😊
[image]

jedgarpark, 2024-09-21:
I can update the file so more/all of the Pico GND pads connect to the ground plane -- right now that is the only one.

DynaMight1124, 2024-09-21:
> Just a side note. I'm not sure you soldered the ground on the pico's debug side. The ground of the sd and hdmi are connected to this. 😊 ![image](https://private-user-images.githubusercontent.com/8100740/369637867-06eab4fd-395a-4343-a8cf-40fe2385199b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5NDMzMTcsIm5iZiI6MTcyNjk0MzAxNywicGF0aCI6Ii84MTAwNzQwLzM2OTYzNzg2Ny0wNmVhYjRmZC0zOTVhLTQzNDMtYThjZi00MGZlMjM4NTE5OWIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MjFUMTgyMzM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDdjYjVkNDNhMmM2ZTUxZjhiMGQ5YjdjMjljYjE1ZWY3NWIzYWEyNzRmODlmOTYxNzA3ODU2NTZlYzIzYzQ2MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.te7vFwkVa_rV4idc36OvTCf_XV9zbKz9JnRTC_Qnfdo)

It's all soldered, just most of it has been wicked away. I built it up with some nice solder blobs, tested it and initially it was all good but after like 5mins it crashed, then kept on crashing. I originally assumed it was a faulty Pico (its a AliExpress cheap one) so started to wick away to hopefully remove it, then I started to wonder if a few pins were bridged underneath. Turned out a few pins were bridged, it works perfectly now and its technically all soldered, just looks pretty bare! by that stage I was 'if it aint broke'

jedgarpark, 2024-09-21:
@DynaMight1124 the new model looks great! Gonna print one right away.

fhoedemakers, 2024-09-21:
I'll also print one.

fhoedemakers, 2024-09-21:
FYI

Sometime ago i ported an emulator for the [Sega Master System and Game Gear](https://github.com/fhoedemakers/pico-smsplus). Works on Pico, better on Pico 2.

Today I released port of a [DMG Game Boy Emulator](https://github.com/fhoedemakers/pico-peanutGB), Pico 2/RP2350 only.

Both not perfect, but a lot of games work.

DynaMight1124, 2024-09-21:
I have been using the MS version on my other one, works really well, I'm a Sega fanboy at heart. I did try to get a Megadrive emulator to work (https://github.com/xrip/pico-megadrive/) but I couldnt get it to compile after changing the pins to match. I doubt it'll run all that well on the RP2040 but when I heard about the RP2350, I started to wonder if it would be viable. 

I'll need to get a RP2350 and give the Gameboy one a try. 

Not even sure if its possible, would more than one emulator be possible? its a 2MB flash, NES and SMS are about 700k each, GB 400k, would a front end to select an emulator be possible?

P.S. let me know if the case works. As far as I know I'm the only one thats ever printed them, not all 3D printers are the same so always worried it wont work for everyone.

fhoedemakers, 2024-09-22:
@DynaMight1124  and @jedgarpark 
I added info about the new design to the README, CHANGELOG and latest Release page. Please let me know if you require any corrections. If  leave this issue open for this reason.

> Not even sure if its possible, would more than one emulator be possible? its a 2MB flash, NES and SMS are about 700k each, GB 400k, would a front end to select an emulator be possible?

I'm playing with this idea for a while know. But not sure how this can be done considering the amount of RAM and flash memory. (Although Pico 2 has more of both)

> I have been using the MS version on my other one, works really well, I'm a Sega fanboy at heart. I did try to get a Megadrive emulator to work (https://github.com/xrip/pico-megadrive/) but I couldnt [...]

Did not know of that one. Will certainly look into this one. But memory constraints and performance will sure weigh in.

DynaMight1124, 2024-09-24:
I'm certainly no coder and have no idea if its likely or possible. I was recently looking at the Odroid Go, which is based on an ESP32, although it does use the one with 16MB flash but that does support NES, GB, GBC, GG, SMS in one with a front end loader. I assume the front end would load each emulator into memory when selected. I dont know if your current front end loads the emulator into memory straight away or it loads when the game is loaded/selected. 

Again, probably totally wrong but if we can use upto 2MB flash to cover 2-3 emulators, maybe more if theres a larger flash (I believe there are Pico1's with larger flash out there too) then only load the emulator when selected, we could have a multi emulator.

fhoedemakers, 2024-09-26:
I'll receive my print today or tomorrow. But I won't be able to test it right away because I have surgery tomorrow and will need some weeks to recover afterward.

About the multi emulator solution. The biggest constraint is the amount of RAM available. I'll see what i can do about this.  But don't hold your breath for this. :-)

DynaMight1124, 2024-09-26:
I hope everything goes well with the surgery.

fhoedemakers, 2024-10-04:
Still recovering from surgery last week. But so now and then i have some time and energy to do some fun things. :slightly_smiling_face:
I received the case last week, and today I assembled the PCB into it. I was impressed how easy this was.

Maybe one suggestion for enhancement. If you could change the top cover to hold a 3D printed button above the Pico's bootsel button. This makes software updates possible without unscrewing the case.  
(I'm working on an update to make the zapper usable. It's kindish working)

[image]

DynaMight1124, 2024-10-04:
Glad you're on the mend and that the case fitted together ok. 

I did think about a bootsel button as I've updated mine a few times and its a little annoying, I ended up wedging something thin and non-conductive between the USB cable and opening to push the button to save the hassle of re-opening :)

The button shouldnt be a problem, it obviously wont look as nice on top, not sure if I can do much to hide it and it'll be way to complex to create some kind of mechanism on the back. I have been wanting to test a different way of a button press built into the design using the flex of the plastic, I had intended to try to on another design I've been working on but then Fusion crashed prior to me saving the initial design and everything is gone! Lesson learnt, name a project straight away! just not sure if I can be arsed to do it all again.

jedgarpark, 2024-10-04:
Since BOOTSEL is broken out as a pin on the Pico i could route the PCB to put a button next to the on off switch.

DynaMight1124, 2024-10-04:
It could be a 90 Degree tactile switch possibly? 

I think I can probably sort it, I did re-design my lost design and added the push button I wanted to try and it works pretty well, been pressing it all evening like some kind of fidget button! Its possible the NES one could be a little trickier as this one is only 3mm from the case to the button whereas the NES would be over 10mm leading to a larger swing but I'll have a play and see how it goes.

[image]

jedgarpark, 2024-10-04:
right on. i've built that type of actuator before, but only when it's very near the switch.

fhoedemakers, 2024-10-05:
Hi @DynaMight1124 and @jedgarpark I'm not sure i understand it correctly. But is it so that on the PCB the bootsel will be redirected to a button/switch  near the on/off switch and the case design will be changed accordingly? Sounds great to me. :+1:

As a side note, I want to suggest another change to the PCB design. Is it possible to connect D3 and D4 on NES controller port 2 to GPIO28 (D3) and GPIO27(D4)?  I'm trying to get the zapper working with the emulator. Should be possible on LCD displays. Got it already kind of working with Duck Hunt.  Not sure if i can work it out successfully,  but i think changing the design does not harm.
(see also https://github.com/fhoedemakers/pico-infonesPlus/discussions/69 )

DynaMight1124, 2024-10-05:
I have added a little button on the top shell to allow the bootsel to be pressed. It does work pretty well, even with the 12mm distance between shell and PCB. It doesnt look terrible imo and its all printed within the top shell (button shell hasnt changed) I have done a few designs with an additional button but it can be fiddly and quite like this design. 

[image]

jedgarpark, 2024-10-05:
> As a side note, I want to suggest another change to the PCB design. Is it possible to connect D3 and D4 on NES controller port 2 to GPIO28 (D3) and GPIO27(D4)? I'm trying to get the zapper working [...]

Yes, I'll make that update. I thought zapper could only work on a CRT, but maybe the emulator uses some trickery to make it possible on LCD?

fhoedemakers, 2024-10-06:
> I have added a little button on the top shell to allow the bootsel to be pressed. It does work pretty well, even with the 12mm distance between shell and PCB. It doesnt look terrible imo and its all [...]

The design in the picture looks good. Let me now when you have updated the Thingiverse page so i can print a new top myself 😃 

Thanks!

fhoedemakers, 2024-10-06:
> Yes, I'll make that update. I thought zapper could only work on a CRT, but maybe the emulator uses some trickery to make it possible on LCD?

Thanks John! 👍 

I also thought it would never work on LCD displays until a user drew my attention to this: https://neslcdmod.com/

You need a hardware modified NES zapper or a cheap unmodified third party one.The site contains patches you have to apply to your zapper roms to compensate for the timings caused by the lag from the LCD.

I already got it working with the 'single duck' mode in Duck Hunt. Not sure if i can completely pull it off though.

DynaMight1124, 2024-10-06:
I have added the design to Thingiverse, I've left the original one there too as people have the option of either top case.

https://www.thingiverse.com/thing:6689537

fhoedemakers, 2024-10-06:
Thanks @DynaMight1124 Will update the README and I just have ordered a print of the top cover.

fhoedemakers, 2024-10-07:
I updated README.md, CHANGELOG.md with info about the additional top cover with button.

jedgarpark, 2024-10-07:
[picones_v2.1.zip](https://github.com/user-attachments/files/17280039/picones_v2.1.zip)
Here's the new version of the PCB with the second controller Zapper connection (GPIO28 (D3) and GPIO27(D4)).

fhoedemakers, 2024-10-22:
Hi @jedgarpark and @DynaMight1124,
I finally had time to test the 2.1 PCB and the push button 3-D top cover. Push button works fine. I also see that John added more ground points to the PCB. That is also nice!
@DynaMight1124 , John updated the AdaFruit blogpost about this project, he also mentionend your 3D-Design. https://learn.adafruit.com/nes-emulator-for-rp2040-dvi-boards/overview

The new release v0.20 of the software adds support for more USB input devices, like XBOX (XInput), Keyboard and more. Will add it soon to the Master System emulator too. 😉 

If it is ok for both of you, i'll close this issue. Thanks again both for your great contribution to this project,

DynaMight1124, 2024-10-24:
Glad the push button worked for you. I did look at one of those light guns, couldnt find any cheap enough but will keep my eye out. Used to love light gun games.

I did see the new updates, they look good. Its amazing how many features you've added. With the extra power of the new Pico2, it could open up more systems too.

I think its safe to close this issue

closed 2024-10-24 by fhoedemakers