PicoPlus-devel/pico-infonesPlus #60: Case for the Waveshare RP2040-PiZero

DynaMight1124, 2024-09-07:
Hi

I've designed a case for the Waveshare RP2040-PiZero. Theres info in the Thingiverse link but you basically need the board, 2x NES Ports, some wires to connect them together and 4x M3 6mm (or 4mm) bolts. Its a little bit of a squeeze for the wiring as it was getting a little out of proportion, not sure if its for the best all the time but I've always tried to keep my models as small as possible to save on print time and plastic.  

Link for the files: https://www.thingiverse.com/thing:6758682

[image]

fhoedemakers, 2024-09-08:
Thank you so much for the design. I've updated the README file with images and the link to your design

closed 2024-09-08 by fhoedemakers

fhoedemakers, 2024-09-08:
I also put a link to your design onto the release page. https://github.com/fhoedemakers/pico-infonesPlus/releases/tag/v0.17

I think it is convenient to put it there. Then all is one one place

PR #61 (2024-09-08): Added links to 3D case design for Waveshare RP2040-PiZero
Added links to 3D case design for Waveshare RP2040-PiZero. Case design done by @DynaMight1124 

Closes #60