PicoPlus-devel/pico-infonesPlus #64: Wii Classic Controller / Waveshare RP2040 PiZero Board

SRDWA, 2024-09-13:
Hello,
I used the Waveshare RP2040 PiZero board for my setup. I've now noticed that the Wii Classic Controller doesn't work. I connected the Adafruit Wii Nunchuck Brackout Adapter to 3.3V, GND, SCL -> GPIO3 and SDA -> GPIO2. Have I made a connection error? Or is the Wii Classic Controller not supported when using the Waveshare board?

Best regards, Sven

fhoedemakers, 2024-09-13:
Hi Sven,

The Wii Classic Controller is not natively supported by the Waveshare device, but there is no reason it couldn't be.

Below  a zipped .uf2 file for the Waveshare device I just built, which adds support for the Wii Classic Controller on the specified GPIO ports. Could you please give it a try and provide feedback on whether it works? I cannot test it myself

[piconesPlus.zip](https://github.com/user-attachments/files/16995225/piconesPlus.zip)

If this works, i'll add it to the official release.

Frank

SRDWA, 2024-09-13:
Hi Frank,
First of all, thank you very much for your quick reply. I am even more grateful to you for the image. The title selection menu and the games work with the Wii Classic Controller and can be operated without any problems.
Would it be too much to ask if you could also provide an image with the pico smsplus? (same problem)

fhoedemakers, 2024-09-13:
Hi Sven,

Will do this tomorrow. No time anymore today. I'll also make a new release for both emulators with these changes.

Frank

SRDWA, 2024-09-13:
Just for information. I use the following cable:
https://www.adafruit.com/product/4397

Here is the connection assignment:

[image]
[image]

You can use the images for the README.

closed 2024-09-13 by SRDWA

fhoedemakers, 2024-09-13:
Thanks, will do. Is that an LCD display on which the waveshare is mounted? Can you share the specs? Looks cool.

fhoedemakers, 2024-09-13:
Below the smsPlus with wii-classic support:

[waveSharePicoSmsPlus.zip](https://github.com/user-attachments/files/16997083/waveSharePicoSmsPlus.zip)

Let me know if this works.

SRDWA, 2024-09-13:
The monitor is a generic 7" display for Raspberry Pi's.
https://www.amazon.de/dp/B0D1C4XLNY?ref=ppx_yo2ov_dt_b_fed_asin_title

Unfortunately the image doesn't work. The RP2040 doesn't produce an image. Even pressing the RUN button doesn't change anything.

SRDWA, 2024-09-13:
Sorry. You should also plug in the HDMI cable.
The image works.

fhoedemakers, 2024-09-14:
Hi Sven,

I just published new releases for InfoNes and SMSPlus.

I updated the (InfoNES) README with details how to setup and I included your pictures.

Let me know if i missed something.

Frank

SRDWA, 2024-09-14:
Good morning Frank.
I can only thank you once again for the great support. I hope that both nespi/pico sms projects continue to be well maintained and that they continue to develop. Great work! 👍👍👍