PicoPlus-devel/pico-infonesPlus #68: Support for another USB Gamepad

A-Ibrahim, 2024-09-24:
Hi, Could you add support for this USB game pad:
Product ID: 0x081f
Vendor ID: 0xe401
[https://www.aliexpress.com/item/32863906297.html](https://www.aliexpress.com/item/32863906297.html)

another question is this one supported?
[https://www.aliexpress.com/item/32895822539.html](https://www.aliexpress.com/item/32895822539.html)

fhoedemakers, 2024-09-26:
I flagged this issue with "help wanted". I don't have theses controllers at hand.

A-Ibrahim, 2024-09-26:
I have the first controller and Waveshare RP2040-PiZero tried to compile debug version the compilation failed
`pico-infonesPlus (main)~ ./bld.sh -d -c 4`

`[ 10%] Built target pioasmBuild
drivers/sdcard/CMakeFiles/sdcard_spi_pio_h.dir/progress.make:1: *** missing separator.  Stop.
make[1]: *** [CMakeFiles/Makefile2:2046: drivers/sdcard/CMakeFiles/sdcard_spi_pio_h.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 10%] Generating dvi_serialiser.pio.h
[ 11%] Built target dvi_dvi_serialiser_pio_h
make: *** [Makefile:91: all] Error 2
`

fhoedemakers, 2024-09-26:
Can you attach the complete output of bld.sh?

A-Ibrahim, 2024-09-26:
do I need debug probe to send you debugging outputs?

fhoedemakers, 2024-09-26:
Debug probe is the best option, but not necessary when you use a  Raspberry PI 4 or 5 for building the project. You then can connect GPIO0 GPIO1 and GND from the Pico to the raspberry Pi and use minicom to capture the debug printf statements. I see they have left it out on the recent getting started guide.

See [this older version of the getting started guide](https://www.mouser.com/pdfDocs/getting-started-with-pico.pdf?srsltid=AfmBOoqXZqHsN6IW3vuzzKng2QIhW42waVG4RV8G5cmaV4lvNdF0KlkM)  chapter 4.5. See "Hello World" UART output.

A-Ibrahim, 2024-10-10:
sorry for the delayed response
`HID device address = 1, instance = 0 is mounted
VID = 081f, PID = e401
HID has 1 reports and interface protocol = 0:None`

fhoedemakers, 2024-10-10:
Ok, thanks. In the next couple of days i will create a testbuild for this controller type. This will not get you a functional controller, but it will output some debug data UART. Like the output you gave me. 
Just tell me which config you are using so in can build it for that config.

Frank

A-Ibrahim, 2024-10-10:
I compiled debug version for Waveshare RP2040-PiZero

fhoedemakers, 2024-10-11:
Can you also send me an image and the brand and product name of the controller?

A-Ibrahim, 2024-10-11:
Does not have any branding name on it.
[https://postimg.cc/fJzZg6M3](https://postimg.cc/fJzZg6M3)
[https://postimg.cc/VJ2x1TPW](https://postimg.cc/VJ2x1TPW)

fhoedemakers, 2024-10-11:
Below a test debug build for the WaveShare device.
It dumps the input received from the controller like this (I tested with a Genesis mini usb controller i had lying around). You should see continuous debug prints like this:

[image]

For your gamepad, the output starts with "MantaPad: len ="

If you don't see this kind output, the gamepad might not work.
If it works can you help me decipher what Hex/Binary byte has changed by each button press? 
The changes where done in hid_app.cpp. You can look it up by changing the branch from main to USB.

[piconesPlus.zip](https://github.com/user-attachments/files/17343869/piconesPlus.zip)

A-Ibrahim, 2024-10-11:
Pulled and compiled USB branch here is the output of the key pressing
```
This without pressing any button:
MantaPad: len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                        7f 7f 00 80 80 0f 00 00
This after keep pressing X blue button:
MantaPad: len = 8  - 01111111 01111111 00000000 10000000 10000000 00011111 00000000 00000000
                        7f 7f 00 80 80 1f
This after keep pressing A red button:
MantaPad: len = 8  - 01111111 01111111 00000000 10000000 10000000 00101111 00000000 00000000
                        7f 7f 00 80 80 2f 00 00
This after keep pressing B yellow button:
MantaPad: len = 8  - 01111111 01111111 00000000 10000000 10000000 01001111 00000000 00000000
                        7f 7f 00 80 80 4f 00 00
This after keep pressing Y green button:
MantaPad: len = 8  - 01111111 01111111 00000000 10000000 10000000 10001111 00000000 00000000
                        7f 7f 00 80 80 8f 00 00
This after keep pressing START button:
MantaPad: len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00100000 00000000
                        7f 7f 00 80 80 0f 20 00
This after keep pressing SELECT button:
MantaPad: len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00010000 00000000
                        7f 7f 00 80 80 0f 10 00
This after keep pressing UP button:
MantaPad: len = 8  - 01111111 00000000 00000000 10000000 10000000 00001111 00000000 00000000
                        7f 00 00 80 80 0f 00 00
This after keep pressing DOWN button:
MantaPad: len = 8  - 11111111 11111111 00000000 10000000 10000000 00001111 00000000 00000000
                        ff ff 00 80 80 0f 00 00
This after keep pressing LEFT button:
MantaPad: len = 8  - 00000000 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                        00 7f 00 80 80 0f 00 00
This after keep pressing RIGHT button:
MantaPad: len = 8  - 11111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                        ff 7f 00 80 80 0f 00 00
This after keep pressing rear LEFT button:
MantaPad: len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000001 00000000
                        7f 7f 00 80 80 0f 01 00
This after keep pressing rear RIGHT button:
MantaPad: len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000010 00000000
                        7f 7f 00 80 80 0f 02 00
```
one more request don't delete USB branch for future game-pads support

fhoedemakers, 2024-10-11:
Great, this controller will certainly work.

Thanks for looking up the button presses. I will continue to work on this and also on the Genesis Mini. I'll keep you informed.

fhoedemakers, 2024-10-12:
Can you try this .uf2 file from the USB branch with your controller?  
[piconesPlus.zip](https://github.com/user-attachments/files/17350048/piconesPlus.zip)

Cannot test it myself. The Genesis Mini 1 controller however works well for me.

A-Ibrahim, 2024-10-13:
It is getting detected but not working properly
 ```
MantaPad detected - device address = 1, instance = 0 is mounted - VID = 081f, PID = e401
HID has 1 reports and interface protocol = 0:None
```
even when I remove the game pad the menu of the roms sometimes shuffle and sometimes not, Like some kind of gpio header interference  I think

fhoedemakers, 2024-10-13:
[piconesPlus.zip](https://github.com/user-attachments/files/17354950/piconesPlus.zip)
I tested on my own Waveshare device and noticed the .uf2 file i sent you was not good. I included a new one  above. (Now builded in release mode)
It can be bad programming from my side. If it still does not work, i will send you a joypad test program. It is difficult to test when i have not the specific controller at hand. I ordered one myself on AliExpress a few days ago. Hope it is the correct one. I don't expect it to arrive within the next few weeks.
Did you connect the power to USB-PIO and the controller to USB (Caused some weird issues in the past, but it works with my Genesis Mini controller), or did you use an USB-C Y-cable? Can you send me a foto of how it is all connected.

fhoedemakers, 2024-10-13:
[usbcontrollertest.zip](https://github.com/user-attachments/files/17355062/usbcontrollertest.zip)
Above the usb test programm, it prints to serial out the buttons pressed on the controller.

[image]

A-Ibrahim, 2024-10-13:
I'm getting key pressing outputs from usbcontrollertest.uf2 only after unplug and replug usb gamepad
```
HID has 1 reports and interface protocol = 0:None
HID device address = 1, instance = 0 is unmounted
HID device address = 1, instance = 0 is mounted
VID = 081f, PID = e401
HID has 1 reports and interface protocol = 0:None
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                              7f       7f       00       80       80       0f       00       00
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00101111 00000000 00000000
                              7f       7f       00       80       80       2f       00       00
A
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                              7f       7f       00       80       80       0f       00       00
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 01001111 00000000 00000000
                              7f       7f       00       80       80       4f       00       00
B
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                              7f       7f       00       80       80       0f       00       00
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00100000 00000000
                              7f       7f       00       80       80       0f       20       00
START
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                              7f       7f       00       80       80       0f       00       00
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00010000 00000000
                              7f       7f       00       80       80       0f       10       00
SELECT
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                              7f       7f       00       80       80       0f       00       00
MantaPad    : len = 8  - 01111111 00000000 00000000 10000000 10000000 00001111 00000000 00000000
                              7f       00       00       80       80       0f       00       00
UP
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                              7f       7f       00       80       80       0f       00       00
MantaPad    : len = 8  - 01111111 11111111 00000000 10000000 10000000 00001111 00000000 00000000
                              7f       ff       00       80       80       0f       00       00
DOWN
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                              7f       7f       00       80       80       0f       00       00
MantaPad    : len = 8  - 11111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                              ff       7f       00       80       80       0f       00       00
RIGHT
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                              7f       7f       00       80       80       0f       00       00
MantaPad    : len = 8  - 00000000 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                              00       7f       00       80       80       0f       00       00
LEFT
MantaPad    : len = 8  - 01111111 01111111 00000000 10000000 10000000 00001111 00000000 00000000
                              7f       7f       00       80       80       0f       00       00
```
the release compilation still flaky and wired sometime work and sometime not, even hung after pressing reset, here is photo of setup [https://postimg.cc/47kstHVs](https://postimg.cc/47kstHVs)

fhoedemakers, 2024-10-13:
Strange, the only difference with the config i have is you have wired serial out to USB.  I use a debug probe

Just for test, did you already unplug the UART to USB pins and try the emulator again? Maybe that is causing the strange behaviour...

[image]

A-Ibrahim, 2024-10-13:
Removed UART pins still not working properly and then unplug USB pad the menu sometimes shuffles on it own
after the I replug USB pad try select any game red screen flash rp2040 restart to roms menu.

A-Ibrahim, 2024-10-13:
Can you activate internel pullup/pulldown  on GPIO used for other controllers?

fhoedemakers, 2024-10-13:
> Can you activate internal pullup/pulldown on GPIOs used for other controllers?

Not that i'm aware of.  

Do you have an USB keyboard to test? 
[piconesPlus.zip](https://github.com/user-attachments/files/17355305/piconesPlus.zip)
This version is compiled to use an usb-keyboard:

 - A: Select (Or Q depending on your kb layout)
  - S: Start
  - Z: B
  - X: A
  - Cursor keys: D-pad

fhoedemakers, 2024-10-13:
I could disable WII-Pad support (GPIO2 and GPIO3) for testing

fhoedemakers, 2024-10-13:
This is the version with GPIO2 and GPIO3 disabled:

[piconesPlus.zip](https://github.com/user-attachments/files/17355331/piconesPlus.zip)

fhoedemakers, 2024-10-13:
Beginning to suspect your Waveshare board has a malfunction, but i want to wait until my AliExpress controller arrives to rule anything out.

A-Ibrahim, 2024-10-13:
After GPIO2 and GPIO3 disabled both my USB-keyboard and game pad  worked well no more random roms menu shuffling  after selecting a game I have to unplug an re plug USB cable of the game pad or keyboard to start working in the game

fhoedemakers, 2024-10-13:
Ok glad it works now somehow for you. It is odd however, i do not have any issue you describe, even with GPIO2 an 3 still enabled. Keyboard works well, USB gamepad works well everytime. Do not need to unplug and re-plug anything. 

Regarding the message. Don't know why it detects two instances of the same  device. I got the same message however.

fhoedemakers, 2024-10-14:
Just a guess. I'm using a 2Amp USB power supply.  Maybe you are using a  power supply with less amps. If so can you try one with minimal 2Amp?

A-Ibrahim, 2024-10-14:
I have good power supply it can handle up to 3Amp.
How I can disable GPIO2 and GPIO3 in my compilation process?

fhoedemakers, 2024-10-14:
In the CMakeLists.txt Line 155 and 156, set value to -1 in stead of 2 and 3. I just don't get it. It works just fine on my board, every time. Waiting for the AliExpress controller to arrive.

    set(WII_SDA "2" CACHE STRING "Select the SDA GPIO pin for Wii Classic controller")
    set(WII_SCL "3" CACHE STRING "Select the SCL GPIO pin for Wii Classic controller")

fhoedemakers, 2024-10-15:
I just purchased second WaveShare device. This also worked without any problem with the WII gpio pins still enabled. Never needed to re-plug the gamepad to get it working in a game. Also switching keyboard and gamepad back and forth gave no problem.

When you build from source, what version of the SDK are you using? I'm building using the latest Pico SDK 2.0.0. I think this has not important however since you have the same issues with the SDK i send you.

Do you have a second WaveShare RP2040-PiZero at hand on which you can test it?

A-Ibrahim, 2024-10-15:
I'm using SDK 2.0.0, And only have one WaveShare RP2040-PiZero right now.

fhoedemakers, 2024-10-17:
Today i got the AlieExpress controllers. I ordered a NES style and a SNES .

The SNES type controller works without any problems, however the NES controller has the same problem as you describe, i need to re-plug the controller after i started a game. However, the shuffling in the menu's does not occur.  

I tested with GP2 and 3 still enabled.

It's strange you have the problem with the SNES type of controller.

After a game is selected, the RP2040 reboots and starts the game, maybe that leaves the (S)NES  controller in somekind of hung state. I'll look into that.

Another weird thing is that B on the NES controller is X on the SNES.

fhoedemakers, 2024-10-20:
I merged the USB branch to main. Keyboard, XInput, Genesis Mini and PSClassic controllers added.
The AliExpress ones are also added, but with a mention in the README and the CHANGELOG that the controller needs to be replugged when starting a game.

A-Ibrahim, 2024-10-20:
Thanks for your effort Mr Frank

fhoedemakers, 2024-10-20:
> Thanks for your effort Mr Frank 

You are welcome.

fhoedemakers, 2024-10-20:
Is it ok to close this issue?

A-Ibrahim, 2024-10-20:
One more last chance please, Is it possible to re-init usb game pad after emulator reboots `tud_disconnect()` and `tud_connect()`
[https://github.com/hathach/tinyusb/issues/428](https://github.com/hathach/tinyusb/issues/428)

fhoedemakers, 2024-10-20:
I think i need some help here for this.