PicoPlus-devel/pico-infonesPlus #71: Add a standard USB Keyboard support

cgchaconr, 2024-10-04:
is more common and cheap than PS4 controllers

fhoedemakers, 2024-10-13:
[piconesPlus.zip](https://github.com/user-attachments/files/17355315/piconesPlus.zip)
Can you test? Please use the correct .uf2 in the above zip-file for your configuration.

- A: Select (Or Q depending on your keyboard layout)
- S: Start
- Z: B
- X: A
- Cursor keys: D-pad

cgchaconr, 2024-10-13:
Hi

fail test, i am suing the Waveshare RP2040-PiZero, load the correct image,
load the rooms and boot but nothing happen with the keyboard

fhoedemakers, 2024-10-14:
Hi,

Sorry, I forgot to mention this earlier:

The PIO-USB cannot be used to connect the USB keyboard because HDMI-DVI and PIO-USB cannot be used simultaneously. [Please refer to the Waveshare product page for more details](https://www.waveshare.com/rp2040-pizero.htm)

However, can you try connecting the keyboard to the port labeled "USB" and try again? While it's not recommended, the PIO-USB can still be used to power the board.  [For more information, see the Waveshare section in the README file](https://github.com/fhoedemakers/pico-infonesPlus#waveshare-rp2040-pizero-development-board)

cgchaconr, 2024-10-14:
That works

However don't load room's

fhoedemakers, 2024-10-14:
To rule out bad roms, can you unzip copy bladebuster.nes from this url to your SD card and run it: http://hlc6502.web.fc2.com/BB_20120301.zip This is  a homebrew shootem-up and is freely distributable. This must certainly work.

fhoedemakers, 2024-10-14:
Also:
Use NTSC roms, not PAL.
Using the menu:
- Press keyboard-x to  load rom or open folder
- Press keyboard-z to go to previous folder
- Press  keyboard-S to start last played game.

cgchaconr, 2024-10-14:
using bladebuster room send error message: cannot open /currentloaderrom.tx

fhoedemakers, 2024-10-14:
Strange. Is it possible to remove settings.dat and /currentloadedrom.txt from the sd card and try again?

cgchaconr, 2024-10-14:
finally works

fhoedemakers, 2024-10-14:
How did you solve it?

cgchaconr, 2024-10-14:
Deleting settings.dat

fhoedemakers, 2024-10-14:
And you can also load your own rims?

cgchaconr, 2024-10-14:
> And you can also load your own roms?

Yes, it works!

PR #79 (2024-10-20): More  USB controllers and USB keyboard added
Closes #71 

See also #68 and #26
closed 2024-10-20 by fhoedemakers