PicoPlus-devel/pico-infonesPlus #76: Red flash and restart when trying to load up a game

basvanoers, 2024-10-12:
hello @fhoedemakers 
i recently finish building your amazing project, everything works  but when i try to start up a game a red screen flashes and the pico restart back to the sd card menu. 

I also bought a new 16gb sd card thinking that was the problem, but it didn't fix anything. 
Does anyone know what might be the issue?

thanks in advance 
Bas

fhoedemakers, 2024-10-12:
Hi Bas,

I assume from your name you are Dutch speaking? 😄 I will continue in English however.

As i understand correctly, the menu is visible and the directory contents are shown. 
This means that either

- the SD card cannot be read
- the rom cannot be written into flash memory because of hardware issues. (not likely)
- The rom is not in the correct format

Can you tell me what config you are using, and whether you built the .uf2 from scratch or downloaded the latest release from the releases page?

basvanoers, 2024-10-12:
Hello @fhoedemakers,
Thank your for the fast reply. 
Funny you noticed my name, I am indeed Dutch.
 I just used the uf2 file that was in your github. I am using a pi pico 2 with a dvi socket from elektor, and a SD card reader from otronic.

Thanks in a advance
Bas

fhoedemakers, 2024-10-12:
Dan zou het ook in het Nederlands kunnen... 😄 

Ok, does the SD card reader need 3 or 5 volt?

basvanoers, 2024-10-12:
I needs 3.3 volt to work

fhoedemakers, 2024-10-12:
To rule out bad roms, can you unzip copy bladebuster.nes from this url to your SD card and run it: http://hlc6502.web.fc2.com/BB_20120301.zip This is  a homebrew shootem-up and is freely distributable. This must certainly work.

The Pico is overclocked in order to keep up with the HDMI display.  I noticed some brands of SD-breakouts don't work well with overclocked Pico's. See #6  Can you consider purchasing  [this one from Kiwi Electronics](https://www.kiwi-electronics.com/nl/microsd-card-breakout-boardplus-3206?search=MicroSD%20card%20breakout%20board%2B) ? 

I assume you are using a breadboard:

- If you soldered headers yourself onto the Pico, please check your solder joints.
- Please check your wiring, are all wires from and to the SD card firmly connected?

As a last resort: The Emulator writes a lot of debug information to UART over GPIO0 and GPIO1, is it possible for you to capture this output?  See [4.5. See "Hello World" UART output](https://people.ece.cornell.edu/land/courses/ece4760/RP2040/Hunter_C_setup/CSetup/Getting_started_c.pdf) from this older version of the Pico C++ getting started guide. You need a Raspberry PI for this however.
You can also use a DebugProbe and a Windows PC to capture the output, but the probe [needs to be purchased ](https://www.kiwi-electronics.com/nl/raspberry-pi-debug-probe-11345?search=debug%20pro)

Hope you get it working.

basvanoers, 2024-10-12:
I tested it with the rom you send, and it worked!!!! So there probebly was something wrong with the rom I tryed. Thank you soo much for helping me.

fhoedemakers, 2024-10-12:
Graag gedaan!
Voor de juiste roms: Zoek in archive.org (wanneer terug online) op htgdb-gamepacks.

closed 2024-10-12 by fhoedemakers