PicoPlus-devel/pico-infonesPlus #102: Intermittent Audio Dropouts

parandandrd, 2024-12-24:
Issue: audio drops out every minute or two for a couple of seconds, then comes back. Gameplay is uninterrupted and there are no other issues.

I've seen this on two different builds now, one using a Pico and another using a PIco 2, both using the Adafruit DVI sock and SD card breakout. I've tried both the ARM and RISC-V builds and the issue still occurs, though the RISC-V seems to be somewhat less affected.

Everything else is working fine and I don't see that anyone else has reported this, so I'm not sure if it's a software issue or something I've been doing wrong in the setup process. Any help would be appreciated!

parandandrd, 2024-12-24:
Update: The previously described behavior was seen on a Sony X800E TV set. I tried connecting the unit to an ASUS monitor with built in speakers and the audio issues disappeared. It's worth noting that in order to even the the board recognized by the SONY I had to wire up the 5V pin on the HDMI sock. It's further worth noting that I've had some experience with this model TV being very picky about HDMI in general.

I'm going to do some more testing with a few more displays I have around and report what I find back here before I close the issue. Perhaps this will of be use to someone else with a insolent Sony TV someday...

fhoedemakers, 2024-12-25:
Thanks for your message. 
I am aware of this; the audio dropouts have also been present on my 2020 Samsung QE55Q95 TV since the beginning, but I never paid much attention to it.

On my capture device, and my cheap AlieExpress LCD display  which i use most of the time, the  dropouts never happen. 
I don't think i can fix this myself. Maybe @shuichitakano who ported the emulator  to the RP2040 has an idea or can help?

The fact that some displays require 5V to be connected to the HDMI breakout was already documented in the README [here](https://github.com/fhoedemakers/pico-infonesPlus#raspberry-pi-pico-or-pico-2-setup-with-adafruit-hardware-and-breadboard) and [here](https://github.com/fhoedemakers/pico-infonesPlus#troubleshooting-no-image-on-tv-or-monitor).

parandandrd, 2024-12-25:
I also noted the issue using the original port by @shuichitakano when building a prototype (again, on the Sony TV). I'm guessing the HDMI audio output is slightly non-standard in some way which most displays cope well with but others do not. I'm unfortunately not familiar enough with HDMI/DVI encoding, timing, etc. to be of much help. I will post results when I have a chance to test with a wider variety of displays. Thank you for all your work on this project!

fhoedemakers, 2024-12-25:
You're welcome. Merry Christmas!

shuichitakano, 2024-12-25:
Hello,
The issue might be caused by a shortcut taken in the audio packet encoding. Specifically, the required Channel Status bits, which include information like the audio sample rate, are not being output correctly.
However, since there are other sources of information related to the sample rate, it seems that even without outputting these bits, the few displays I had on hand for testing worked without any issues.

fhoedemakers, 2024-12-25:
Thanks for your answer, much appreciated! It think the issue primarily occurs on consumer television sets.