PicoPlus-devel/pico-infonesPlus #103: Control issue with v0.23 & WS PiZero

DynaMight1124, 2024-12-26:
Hi

Thanks for the continued improvements, just updated my PicoNES's to 0.23, both of the PCB versions (0.1 & 2.0) worked fine, however when I updated my WS PiZero, the controls didnt work, well not properly. I could press start if I plugged into player 2 but couldnt use any direction buttons. I am using an aftermarket NES controller (original connectors etc, not USB) I havent tested USB controllers but happy to test if that helps. 

I flashed back to 0.22 and the controls are working fine. I assume it is related to the latest update with the CLK high/low. Not sure why the WS board is different, the pins of the NES connector are connected directly to the GPIO on the WS board.

fhoedemakers, 2024-12-26:
Hi, I hope you are doing well,

I just released v0.24, both my NES controllers work fine on this release. I use the orignal ones. Can you install that last version and let me know?

Frank

DynaMight1124, 2024-12-26:
Hi, I have updated to v0.24 and its the same on the WS PiZero. All works fine on the PCB version.

fhoedemakers, 2024-12-26:
Strange. You don't have an original NES controller to test with? Just to rule out the aftermarket controller. It all works fine with my WaveShare PiZero with original controllers.  Just to be sure, can you also test with an USB controller on the WavsShare as you asked before?

fhoedemakers, 2024-12-27:
Just an additional thought. Is VCC of the nes controller port perhaps connected to 3V?
I had issues getting the NES controller to work with 3V on the WSPiZero, so i connected VCC to 5V and the controller started to work.
This is documented in the README.

fhoedemakers, 2024-12-27:
> Just an additional thought. Is VCC of the nes controller port perhaps connected to 3V? I had issues getting the NES controller to work with 3V on the WSPiZero, so i connected VCC to 5V and the [...]

I see from the picture of your 3D print in the README the ports connected to 5V. So this cannot be the problem.

fhoedemakers, 2024-12-27:
Hi,
I did some more testing with my Rp2040 pizero board and i after a while my nes controller is starting to  behave erratically.  Maybe it is related. 
I reverted the nes-controller code back to the original one when the WaveShare board is used. Attached a test build. Can you see if it now works for you. If this works, i'll contact the person who changed the controller code.
[piconesPlus.zip](https://github.com/user-attachments/files/18259618/piconesPlus.zip)

DynaMight1124, 2024-12-27:
Hi, sorry for the lack of replies yesterday, I did try to find a USB dongle to plug in a USB controller too but I couldnt find one to test. Yeah it is connected to 5v, I did double check just to be safe.

I have flashed the test build and it works perfectly. Played for about 15mins, multiple games and didnt miss a beat. Not sure why the WS board would differ to the PCB, maybe some additional circuitry between the GPIO and Pico that doesnt exist on the original which is changing the read value? 

Also, happy to raise a new ticket but I had some possible improvements to the menu system if thats ok? Nothing major just a quick scroll when holding the down/up buttons, I know you can press left/right to skip pages but would also be nice to quick scroll when press&holding up/down too, usually this has 2 speeds, slow scrolling to start then if held for a certain time it'll scroll faster. 

Also something else that may help would be going back to A when you get to Z (or back to the start after the last item and visa versa) just makes scrolling through large lists easier.

fhoedemakers, 2024-12-27:
No problem. Glad it works.

For the requested changes  i created issue [#7](https://github.com/fhoedemakers/pico_shared/issues/7))  in the https://github.com/fhoedemakers/pico_shared repostory. This is a shared module also used by the smsplus and Game Boy emulator. This makes it easier to maintain changes in the menu accross emulators.

I will also transfer this issue to that repository since de Nespad code is also in there. I will also contact the person who changed the controller code to look into this.  The code was changed because he designed a PCB with [BlueRetro](https://www.blue-retro.com/) support. For this to work, the nespad controller needed to be changed.