PicoPlus-devel/pico-infonesPlus #105: HDMI problems + Features ADD

vslinuxdotnet, 2024-12-26:
Hello,
Congratulations for this awesome project!
I build it and detect some things to be fixed/add.

I tested in some televisions adapter/converters and all works excepts some issues in T.SK105A 03  board.
The T.SK105A is a board that is used to get a second life to displays without motherboard, in this case i get signal only when the game is running, and get lost when the games get more then 3seconds to load.
At first boot no image is displayed, says no hdmi signal but when a press A in keypad goes to the last game and works.
If i go back to menu, it works always, except in some bad game or if i press the PICO reset button.
It seams that if get lost the hdmi sync never gets back on. May some issue with HDMI timing ?

The others things are, create some menu to change options like resolutions (available by buttons in keypad).
Add a option to load the last file at boot time.
Add some graphic in the display to wait the load into memory, without the red screen with some menu flicks.

Thank you,

VS

fhoedemakers, 2024-12-26:
Hi,

Thanks for your message. 

I don't know that type of board and i'm not sure it will ever work. I did not write the dvi-driver myself. It shares the DVI driver (pico_lib) from the original creator of the port. https://github.com/shuichitakano/pico-infones All i did was making it more usable by adding sd-card support, a menu, two player support, extra controller support etc...
There is always a reboot when switching from menu to game. This initializes the DVI-driver.

The resolution can only be changed in-game and is saved into a settings file so it will be remembered during reboot. The menu uses a fixed resolution of 320x240 with 40x30 text columns/rows.  Have no plans to change this. There are some key combo's that allow the menu colors to be changed and saved. (But i assume you already knew this)

To start the last loaded game, simply press start in the menu.

To show somekind of progress on  the display during loading from sd to flash memory is technically  not possible.  I had to spend a lot of time making this even to work like it is now.  When starting a game, the board reboots and writes the game to flash before the display driver is initialized. This was the only way to make this work reliable, without locking up or display a lot of red flicker.

The red flicker also occurs when reading the SD directory contents for the menu. This cannot be avoided, because the hdmi output must be fed at a constant rate of 60fps which is impossible to achieve when the sd card is read.

vslinuxdotnet, 2024-12-26:
Hi again,
the resolution 320x240 is only at boot for start menu ?
The 640x480@60 works for the games, but when press: select+start they back to menu as 320x240 or continues as 640x480 ?
when is in game, if i disconnect the HDMI cable and put in back the display works again, but if i do the same in the menu, never get back image in display, what  is the difference ?

fhoedemakers, 2024-12-26:
Both the menu and game are rendered in 320x240, which is then  upscaled to 640x480.  The menu uses the same screen settings as one of the in game settings you can change with select-up/down (widest with no scanlines)

fhoedemakers, 2024-12-26:
What comes to mind is this: What configuration are you using? If using a breadboard , you must connect the via labeled 5 on the side of the hdmi breakout board to 5V (VBUS), when using the PCB, make sure this via is soldered with header pins to to PCB.

vslinuxdotnet, 2024-12-26:
Thank you for the interest.
It's a custom board that i quick made, but i have the 5v in pin 18, and also ground the 17.

[image]

All the resistors in place.

[image]

I think my hardware setup is ok, seams something with HDMI signal at startup.

With HDMI->VGA  adapter works fine

fhoedemakers, 2024-12-26:
So 5V is connected to pin 40 of the pico (right to USB)? I cannot visually verify this

fhoedemakers, 2024-12-26:
I see you are using usb c for power. Think 5V must be connected there

fhoedemakers, 2024-12-26:
Nice work btw! 👍

vslinuxdotnet, 2024-12-28:
The 5v is present, is not the HDMI to VGA will not work, the problem is not 5v related.
Is some timing detection issue in HDMI with this screen when pico resets and loads game it works, in first start no video shows until games starts.

fhoedemakers, 2024-12-29:
I don't think i can help you any further with this. As far as i know, you are the first who has display troubles. Since you confirmed it worked on all other displays you tried, it must be related to that T.SK105A 03 board. 

Maybe you can look into the code yourself?

Did you try the original port, has this the same issues? https://github.com/shuichitakano/pico-infones My repo is based on this and uses the exact same dvi_driver https://github.com/shuichitakano/pico_lib

Did you also try to run some older releases v0.20 or v0.19?

closed 2025-10-26 by fhoedemakers