PicoPlus-devel/pico-infonesPlus #110: Why the "Keepout" warning on the PCB?

nsinav, 2025-02-24:
Hello. Thank you very much for this nice project, I have been anxiously waiting for the components to be able to assemble two and thus be able to give one to my brother.

Unfortunately I cannot get both units to work. I have very carefully soldered the Raspberry and all the cards in this project and I cannot get it to work.

I have tested with a tester and the ground is fine, the 5V goes through the VBus pin of the Raspberry Pi Pico and through pin 5 of the DVI breakout, the 14 pins of the DVI Breakout are soldered, the 9 pins of the card for the micro SD and all the contacts of the Raspberry including the 3 rear pins, of which the one in the middle is ground and it is OK.

I am using a Chinese 2GB micro SD card but of good quality, a good quality micro SD cable and I have used 3 different 5V chargers, the Raspberry Pi Pico lights up the LED when I turn on the project with the power switch but it does not give an image at all.

Why the "Keepout" label on the PCB? I have that doubt.

Thanks for your help.

fhoedemakers, 2025-02-25:
Hi,

Not sure what the label means. It is related to the test points on the back of the Pico.  But is not related to any problem at all.

You also mentioned you got no image. Is this solved because you edited this post?

No image can mean you used the wrong .uf2 file. Make sure you flash [pico_piconesPlusAdaFruitDVISD.uf2](https://github.com/fhoedemakers/pico-infonesPlus/releases/download/v0.25/pico_piconesPlusAdaFruitDVISD.uf2)  from the releases page. (Assuming you are using the original Pico not the Pico2)

nsinav, 2025-02-25:
Hi! Thanks for your reply. I modified the question as the problem was that I was flashing the wrong firmware, but I have now fixed it. Very grateful for your reply. Kind regards.

closed 2025-02-25 by nsinav