PicoPlus-devel/pico-infonesPlus #111: Missing Hit Sound Effects in Battletoads and Battletoads & Double Dragon

Pelip, 2025-03-29:
Hi! First off, thanks for this amazing project. I’m running picoNES on a Raspberry Pi Pico 2 with the Adafruit DVI breakout and SD card setup using the pico2_piconesPlusAdaFruitDVISD.uf2 firmware and the v2.1 custom PCB.

I checked most of my games, and they work fine but both version of Battletoads and Battletoads & Double Dragon missing hit sound effect (When you hit enemis). 

Suspected Cause: These two games maybe use the DMC audio channel (sample playback) for hit sounds. I dont know if DPCM playback is implemented here or not functioning correctly in these specific cases.

fhoedemakers, 2025-03-31:
Hi, thanks! I just returned from a two week holiday, so I’m only able to answer now.
I did not write the emulator myself, but used this port : https://github.com/shuichitakano/pico-infones 
Using that port i added sd-card,  rom-selection menu and other features. I’m not familiar with sound implementation nor am i an emulator developer, so i’m afraid I will need help with this to fix.

DynaMight1124, 2025-04-19:
It doesnt look like the original Pico InfoNES developer has done much work on it in quite a few years, you could obviously raise a Issue on their Github. However it does look like the Pico version was ported from https://github.com/jay-kumogata/InfoNES which hasnt been updated in about 7 years

It does look like some developers have used the same emulator and have added some features/bugfixes, but it might be either very hard or very time consuming for these changes to be ported to this emulator. Checkout: https://github.com/xrip/pico-nes Not sure if it covers the sound issue but there does appear to be additional mappers added which does have a few open issues here, I've encountered a few roms that dont work for me either due to a missing mapper (5 I think?)