PicoPlus-devel/pico-infonesPlus #112: I hope the project can add a Turbo function

Hifly-rc, 2025-04-11:
@Author, I have a great interest in the pico-infonesPlus project. I tested the HDMI output quality on a temporary RP2040 development board and was very satisfied. Therefore, I specially designed a development board integrating PICO NES. Everything went smoothly and was great. The only thing I currently hope the project can add is a Turbo function. For example, pressing the SELECT + A keys could activate the A key Turbo, and pressing the combination keys again would deactivate the A key Turbo.

[image]

Hifly-rc, 2025-04-11:
The SFC controller lacks a physical Turbo function. For games that are accustomed to continuous firing, the absence of Turbo can negatively impact the gaming experience. Is it possible to automatically construct Turbo by recognizing the DATA0 of the XY buttons on the controller through software? Additionally, I have another question: How can I enable the game list to display Chinese game ROMs?

Hifly-rc, 2025-04-11:
> There already is a rapid fire toggle in the emulator programmed by the original author of the port: Select + A toggles rapid fire on A Select + B toggles rapid fire on B This is this not working for you?
> 
Thank you for your response. The Turbo feature works as you described!

> Nice design.
> 
I will also share the hardware production materials. However, it is a bit regrettable that the production materials are created based on Chinese language information. If anyone needs it, I am willing to reorganize or submit a PR (Pull Request).
Temporary download path: http://hiflyrc.ip-ddns.com:88/game/Diy-NES/

fhoedemakers, 2025-04-11:
The emulator also supports two controller ports for two player games.  I see your design has only one controller port.

Hifly-rc, 2025-04-11:
> The emulator also supports two controller ports for two player games. I see your design has only one controller port.

Yes, it is built based on the generic Pi3 case. Due to the size limitations of the SFC controller port, only one can be installed. If it were changed to an NES controller, two ports could fit. I don't have an NES controller on hand, so I haven't modeled a dual-port version for the NES controller.
The circuit has added a DC input, which makes it more convenient to support power supply for Type-C USB wired controllers. I have considered creating a version with dual NES controller ports, and the next step is to model the NES port.

fhoedemakers, 2025-04-11:
Oh great! Happy to see that it inspires so many people to come with their own design.
BTW, your board design also supports two player games. The controller connected to the usb port is P1, the (S)NES port is P2
The README has a list of USB controllers that work with the emulator.

closed 2025-04-11 by fhoedemakers